9elements/converged-security-suite #3: Ship dependencies

insomniacslk, 2019-07-14:
Instead of suggesting to run `dep ensure` after cloning the repository, it would be better to vendor them. E.g. the build fails for missing features in google/go-tpm, but after `dep ensure` it correctly points to 9elements/go-tpm (this is also tricky, because the import path is github.com/google/go-tpm but the source is github.com/9elements/go-tpm, https://github.com/9elements/txt-suite/blob/master/Gopkg.lock#L20 ).

flanfly, 2019-07-17:
I see. I'm still a Go novice but if this is The Go Way to do Dependencies :tm: I'll check in the `vendor/` directory.

insomniacslk, 2019-07-17:
the new ways is to use versioned modules, but since I see this project uses `dep` I thought you'd rather use the `vendor` dir. If instead there's no bias against modules, here's the starting point, https://github.com/golang/go/wiki/Modules

hugelgupf, 2019-07-17:
Id suggest just migrating away from dep to modules?

insomniacslk, 2019-07-17:
Yeah my point too. Dep should not be used for new projects

flanfly, 2019-07-29:
I committed the `vendor/` directory to the project. I still think we should move to go modules ASAP.  #38.

closed 2019-07-29 by flanfly