9elements/converged-security-suite #4: Cache TXT support information locally

insomniacslk, 2019-07-14:
The function `ArchitectureTXTSupport` [calls Intel's API](https://github.com/9elements/txt-suite/blob/master/pkg/api/ark-intel.go#L28) to get this information. This may be an issue with environments without connectivity or where external connectivity is not allowed. A possible solution may be caching this information and let the user decide whether to use the cache or the external API.

flanfly, 2019-07-17:
I see the issue, but caching would only apply to the exact CPU model. So in order to make this work you'd to know what CPU(s) the suite will be run and tell the suite to fetch and cache the result for those. I don't think there is a way to simply dump the whole Intel database. If this works for you I'll implement that.

insomniacslk, 2019-07-17:
ideally you would have to know the CPUs that run in your fleet, and pre-cache all of them. If one is missing, the tool should just fail and say that the information is missing

tfg13, 2019-07-30:
I had a brief look, according to https://odata.intel.com/ this entire API is more or less deprecated and we should be using something called "Intel Product Data microservice" - which I wasn't able to find anywhere.

Is it really impossible to query this from the platform we are running on somehow?

insomniacslk, 2019-07-30:
@tfg13 and I are looking for a way to get this information offline, from the running host itself

zaolin, 2019-07-30:
@tfg13 let's just disable this check and print out a warning if internet connectivity isn't foun

insomniacslk, 2019-07-30:
rather than trying and timing out (which may trigger unwanted alerts) better to have rules to include/exclude checks

zaolin, 2019-07-30:
fine for me as well

insomniacslk, 2019-07-30:
also checked with @marcoguerri and we are not sure that this check is useful at all. Also no other test depends on it, so I'm wondering if `test03cpusupportstxt` could be removed entirely

zaolin, 2019-07-30:
@insomniacslk it is useful otherwise you can't verify if the hardware straps and the right cpu are used on the platform

insomniacslk, 2019-07-31:
@marcoguerri is looking into it - we are hoping to infer this information locally rather than having to gather it from an outside environment

PatrickRudolph, 2019-08-01:
Added a black and whitelist in a8fb681a83253fa8f677cbb72abda3c4944d298e
It was generated by script parsing Intel ARK and it works on the Wedge100s.

marcoguerri, 2019-08-01:
I have been having a look without much success. I am curious, is there really no other way (CPUID, MSR registers, etc.) to tell whether the CPU supports TXT, but it's disabled in the firmware settings? 

For example, I have disabled TXT on a machine and rebooted and running txt-stat still shows that SEXIT was done. That information is cached/retained across reboot with TXT disabled, so I guess it must be exposed from the hardware somehow. I was wondering if that could help in this case, but I need to do more study, unless you can confirm/deny straight way.

insomniacslk, 2019-08-01:
> Added a black and whitelist in [a8fb681](https://github.com/9elements/txt-suite/commit/a8fb681a83253fa8f677cbb72abda3c4944d298e)
> It was generated by script parsing Intel ARK and it works on the Wedge100s.

Lovely. I could run it on a disconnected machine :) Thanks!

insomniacslk, 2019-08-01:
I guess this can be closed now

closed 2019-08-01 by insomniacslk