9elements/converged-security-suite #5: Check if MLE implements all interrupts

orangecms, 2019-07-19:
To preserve integrity, an MLE must implement all interrupt handlers. For that purpose, TXT suite can look at the IDT.

See the TXT Development Guide, 2.4.8.

ChriMarMe, 2019-09-11:
We can't verify that with txt-suite.

closed 2019-09-11 by ChriMarMe

orangecms, 2019-09-11:
I see. Chipsec has a kernel module to perform the check, which currently has [an issue](https://github.com/chipsec/chipsec/issues/608) (I just tried it myself and can confirm it). I skimmed through https://www.scitepress.org/papers/2017/61656/61656.pdf to see how it would work and thought it would be rather easy that way. I could imagine there being something like a kernel sanity check on boot to cover that.

Just for clarification: Do you intend to state what txt-suite does/will not cover so people know that they would need additional utilities for those, like Chipsec?