9elements/converged-security-suite #127: Implement a configuration mechanism to provide configuration parameters to tests

marcoguerri, 2020-07-17:
I was thinking that some tests might need to be given input parameters that cannot be hardcoded. For example, if I wanted to write a test which tests that some NVRAM indexes are defined, and protected by a specific Auth Policy, I would probably want to define a list of those indexes and associated hashes in some form of configuration input. A test for NVRAM indexes would read and apply that configuration (In this suite we know we want to test for PS and AUX index, but we don't test the policy for example, which might be relevant if we want to validate that a set of machine is provisioned according to a set of requirements).

Now, the risk is always to make the test non deterministic and too generic, but if there is a clear API to declare which input parameter each test accepts, and the set of parameters is documented, I think the benefits would outweigh the risks. Would this be a feature you would consider?

ChriMarMe, 2020-07-17:
I'm uncertain about that, but I understand your intention.
All tests reflect some part of the specification, that's why all of it is hardcoded. Using indices other than PS, AUX, PO and SGX is up to the user / platform manufacturer / platform supplier and out of scope of TXT spec. 

What else is customizable when setting up TXT other than the AuthPolicy for PS, PO and SGX index?

walterchris, 2020-07-17:
do you have a design idea in mind? Something that I could think of would be having a .yaml file that describes the input and expected response - each test already has a test description where a parameter could be added if it takes input or not. In the yaml file, we would need the test name as an identifier and inputs/expected responses.

This yields, of course, to make it more complex than it has to.

marcoguerri, 2020-07-17:
> All tests reflect some part of the specification, that's why all of it is hardcoded. Using indices other than PS, AUX, PO and SGX is up to the user / platform manufacturer / platform supplier and out of scope of TXT spec.

Agree, it depends on what we thing the scope of the suite should be. My knowledge of TXT is limited, but off the top of my head, LCP policy is something that one might also want to test still within the boundaries of TXT configuration. 

> do you have a design idea in mind? 
I do have some ideas, and it more or less matches with what you mention. Making it part of the test description sounds reasonable. Than you would have param object that exposes to the test its own configuration parameters. Validation of those parameters would be done by the test and maybe one possible outcome of the test could be "invalid parameter", which is a strong signal to indicate that the test did not give any meaningful result.

> This yields, of course, to make it more complex than it has to.

It's more complex, yes, but can be made robust and discoverable. Certainly, if we don't foresee any other use case other than indexes configurations, it's becomes hard to justify this feature.

zaolin, 2020-08-20:
@marcoguerri done https://github.com/9elements/converged-security-suite/tree/master/cmd/txt-suite

closed 2020-08-20 by zaolin