9elements/converged-security-suite #129: IOMMU test failure is not critical if DPR are configured

marcoguerri, 2020-07-17:
Following https://lore.kernel.org/patchwork/patch/783583/, TBOOT might run with IOMMU disabled. @PatrickRudolph explained that there is an alternative mechanism with DPR  protecting memory mapped regions. The test should be amended accordingly.

PatrickRudolph, 2020-07-20:
See Document 315168 Chapter 2.2.4.2 for details

ChriMarMe, 2020-07-20:
After looking through the specs with @PatrickRudolph I came to the following conclusion:
1. DPR is always configured. No DPR -> No TXT.
2. If IOMMU is deactivated, the MLE must be placed in a DPR memory range. 

I'm not quite sure what part is responsible for that. I suspect TBoot. In case it is responsible: 
This can't be checked, because TBoot has not launched a MLE at execution time of the tests. So I have no access to memory addresses for checking this.

PatrickRudolph, 2020-07-20:
You can still run this test from within the MLE, can't you?

ChriMarMe, 2020-07-21:
You do not want tests in the txt-suite which require a launched MLE. 

1. The test should state if TXT support is valid, obviously it its, if a MLE is launched

ChriMarMe, 2020-07-21:
Upsi. Sorry

ChriMarMe, 2020-07-21:
#135

closed 2020-08-03 by ChriMarMe