9elements/converged-security-suite #137: Remove support for WARN

PatrickRudolph, 2020-07-21:
The only reason to have WARN is that some tests are not fully implemented.
Fix all of them and remove the WARN error message.

zaolin, 2020-07-22:
Please elaborate on that here. I don't think it's true

ChriMarMe, 2020-07-23:
The idea behind WARN was to indicate 'optional' configuration which may fail, but not vital for TXT.

PatrickRudolph, 2020-07-23:
> @PatrickRudolph Please elaborate on that here. I don't think it's true. Which tests and what isn't fully implemented?

All tests that are marked NonCritical: true.

> The idea behind WARN was to indicate 'optional' configuration which may fail, but not vital for TXT.

If it's not vital, why would you test it then?

Reading the comment above WARN:
`// ResultWarn indicates that the test failed for the standard configuration but can still be valid in a different configuration of TXT
`
I wonder what the standard configuration is?
Should we list at least all non standard configurations on WARN until the test is able to detect them and return FAIL or PASS?
Something like

Test A: WARN
 * This can happend when .... is configured.
 * This can happend when .... is not present.
 ...

ChriMarMe, 2020-07-23:
> If it's not vital, why would you test it then? 
You may want to check if the PO index is configured correctly, or if a Policy is in PO index. If check passes, everything is ok. If not, it is not vital. 

Different story for PS index. If this one is configured wrong, or the policy is not valid you miss the minimum requirements for running TXT

PatrickRudolph, 2020-07-23:
Simplest fix would be to let the user specify what the "standard" configuration is. It must be specified as command line argument. Knowing that there's no optional configuration any more.

A configuration can be one of:

- Developer mode
- TXT Ready
- TXT FIT Boot HASH autopromotion
- TXT FIT Boot policy
- CbnT
- Running in MLE

closed 2020-08-20 by zaolin