9elements/converged-security-suite #160: provide v2 branch or subdirectory

jensdrenhaus, 2020-09-22:
Since there are tags like v2.x.x, I guess there must be a separate branch or subdirectory for the different major versions.
This is important for the import-paths / module-paths. I cannot include `github.com/9elements/converged-security-suite/pkg/hwapi` for example. Taking this import path Go will take the latest version from 0.x.x to 1.x.x, wich is v1.0.0 here. But at the commit taged v1.0.0  pkg/hwapi is not present yet. -> Failure!

The correct import path would be `github.com/9elements/converged-security-suite/v2/pkg/hwapi`, but this requires a branch or subdirectory associated with v2 including a own `go.mod` file with `/v2` in the module name (it's actually a new module, since in Go major versions do not need to be backwards compatible).

See
* (short solution) https://stackoverflow.com/questions/53344471/taking-a-repository-to-v2
* (detailed info) https://blog.golang.org/v2-go-modules

jensdrenhaus, 2020-09-22:
This will cause quite a bit of work to set everything up correctly now.
An alternative may be renaming the tags as follows:
rename tags v1.x.x in v0.x.x
rename tags v2.x.x in v1.x.x
Of course we need to make sure no one else is using the API at the moment (but I think there is no one yet)

hugelgupf, 2020-09-22:
You do not need a branch or subdirectory.

You only need to put /v2 in the go.mod file, and to change every import path within the repo to use /v2. That's it. No branch or subdir is necessary.

ChriMarMe, 2020-09-23:
We agreed on getting rid of tags at all and move the states of the tags into new branches with the name of the old tags.

Go should get the lastest state of master now, if someone pulls a package and checks out the commit of that state.

closed 2020-10-23 by zaolin