9elements/converged-security-suite #194: build(pcr0tool): Fix building of pcr0tool

xaionaro, 2021-03-26:
```
xaionaro@void:~/go/src/github.com/9elements/converged-security-suite$ go get ./cmd/pcr0tool/
# github.com/9elements/converged-security-suite/v2/pkg/uefi
pkg/uefi/uefi.go:30:2: undefined: uefi.ReadOnly
pkg/uefi/uefi.go:31:2: undefined: uefi.DisableDecompression
# github.com/9elements/converged-security-suite/v2/cmd/pcr0tool/commands/dumpregisters
cmd/pcr0tool/commands/dumpregisters/command.go:60:29: txtAPI.GetMSRRegisters undefined (type hwapi.APIInterfaces has no field or method GetMSRRegisters)
```
The `uefi.*` problems caused by old revision of `fiano` in `go.mod`. Other problems: not fresh enough `pkg/`

ChriMarMe, 2021-03-29:
#198 updates fiano to v6.0.0-rc but this is not the only issue. Commits are missing.

closed 2021-04-07 by xaionaro

xaionaro, 2021-04-07:
Fixed in: https://github.com/9elements/converged-security-suite/pull/209