9elements/converged-security-suite #205: bg-prov template: Contains null

PatrickRudolph, 2021-03-30:
Instead of arrays and strings the template contains JSON _null_ from which it's impossible to infer the type. Fill in "" or [] instead.

ChriMarMe, 2021-04-01:
This argument is false, because a empty array or quationmarks would not help to give information about the type used.

PatrickRudolph, 2021-04-01:
The JSON marshaler uses a string for HashStructure, hs_Buffer, which is a []byte.
The HashList, List field is a slice but JSON still uses [].

ChriMarMe, 2021-04-01:
null != "null"

ChriMarMe, 2021-04-01:
I dont get it.

PatrickRudolph, 2021-04-01:
The main culprit seems to be that the object contains nil slices, but it should contain empty slices.

ChriMarMe, 2021-04-01:
Can we adapt the codegen to solve this issue, @xaionaro ?

xaionaro, 2021-04-01:
> The main culprit seems to be that the object contains nil slices, but it should contain empty slices.

Could you please provide examples of what do we have now and do you want to achieve? Could you also elaborate on the end reason of why the change is desirable (for me to better understand what to do)? Just we know field types using their paths. And there's no neat way to fully preserve typing in JSON (and I do not understand the positive impact of changing `nil` to `[]`).

> Can we adapt the codegen to solve this issue, @xaionaro ?

Looks like so. But I need to better understand what exactly do we want, first :)

ChriMarMe, 2021-04-06:
For example take `HashStructure` 
```
type HashStructure struct {
	HashAlg    Algorithm `default:"0x10" json:"hsAlg"`
	HashBuffer []byte    `json:"hsBuffer"`
}
```
HashStructure should be generated by New()-function and the field `HashBuffer` should be initialized by 
```
HashBuffer := make([]byte,0)
```

ChriMarMe, 2021-04-06:
Same for
```
// HashList describes multiple digests
type HashList struct {
	Size uint16          `rehashValue:"TotalSize()" json:"hlSize"`
	List []HashStructure `json:"hlList"`
}
```
Initialize List by
```
List := make([]HashStructure,0)
```

ChriMarMe, 2021-05-03:
Makes no difference if there is nil, null or [] in the fields. There is no benefit for the reader to understand the field or get information about it's type as long as the field appears in the config.

closed 2021-05-03 by ChriMarMe