9elements/converged-security-suite #268: Fails to build on ppc64le

paulmenzel, 2021-07-26:
With Ubuntu 21.04 on ppc64le, go terminates with a segmentation fault:

    $ GO111MODULE=off go get github.com/tjfoc/gmsm/sm2 github.com/dustin/go-humanize github.com/linuxboot/fiano/pkg/uefi && go run ./pkg/intel/metadata/manifest/common/manifestcodegen/cmd/manifestcodegen -check ./pkg/intel/metadata/manifest ./pkg/intel/metadata/manifest/bootpolicy ./pkg/intel/metadata/manifest/key
    go build github.com/dustin/go-humanize: /usr/lib/go-1.16/pkg/tool/linux_ppc64le/compile: signal: segmentation fault
    go build github.com/linuxboot/fiano/pkg/uefi: /usr/lib/go-1.16/pkg/tool/linux_ppc64le/compile: signal: segmentation fault

xaionaro, 2021-07-26:
@paulmenzel Thank you for the bugreport. But it looks like the problem is inside the Go compiler. Nevertheless I'll try to investigate. Could you recommend an instruction how to run a PPC64LE Ubuntu image in KVM?

paulmenzel, 2021-07-26:
You should be able to able to use [the 21.04 ISO][1] and run it with

    qemu-system-ppc64le -M pseries -m 4G -hda ubuntu-21.04-live-server-ppc64el.iso

[1]: https://cdimage.ubuntu.com/releases/21.04/release/ubuntu-21.04-live-server-ppc64el.iso

xaionaro, 2021-07-26:
Yeah, I also saw similar command on stack overflow, but it does not work for 21.04 (`qemu` just unexpectedly exits while the boot) on my machine. Meanwhile will try 20.04, instead.

xaionaro, 2021-07-26:
@paulmenzel : Which exactly tool are you trying to build?

paulmenzel, 2021-07-27:
I am testing the coreboot toolchain on ppc64le. The sample configuration for cbnt `config/config.ocp_deltalake_cbnt` fails to build.

To reproduce, I just changed to the repository, and executed the lines from `.circleci/config.yml`.

    $ go version
    go version go1.16.2 linux/ppc64le
    $ apt-cache policy golang-go
    golang-go:
      Installed: 2:1.16~0ubuntu1
      Candidate: 2:1.16~0ubuntu1
      Version table:
     *** 2:1.16~0ubuntu1 500
            500 http://de.ports.ubuntu.com/ubuntu-ports hirsute/main ppc64el Packages
            100 /var/lib/dpkg/status

xaionaro, 2021-07-27:
@paulmenzel : Unfortunately, it takes too much time to reproduce the problem (because 21.04 does not work, and on other versions the problem seems to be non-reproducible). Can we wait until https://github.com/golang/go/issues/46858 fix will be merged? (I guess it will be in Go1.16.7)

paulmenzel, 2021-07-28:
Yes, we can definitely wait. Thank you very much for looking into this so quickly.

paulmenzel, 2021-07-29:
[Also fixed by Michael’s Linux patch.](https://lore.kernel.org/linuxppc-dev/99ce2761-ba84-72a3-caaa-5281296fdbc7@molgen.mpg.de/T/#mf8eabb90e29e132ed974afcccbffb2da6b2fa5ac)

As the problem is not in the converged-security-suite, I am closing the issue.

closed 2021-07-29 by paulmenzel