9elements/converged-security-suite #269: bug(hwapi): Fails to build on ppc64le: type TxtAPI has no field or method ProcessorBrandName

xaionaro, 2021-07-26:
PPC64LE, Ubuntu 18.04, Go1.16.6

```
$ go get github.com/9elements/converged-security-suite/v2/cmd/pcr0tool@master
# github.com/9elements/converged-security-suite/v2/pkg/hwapi
/home/xaionaro/go/pkg/mod/github.com/9elements/converged-security-suite/v2@v2.4.1-0.20210719144404-92b4758a5973/pkg/hwapi/api.go:71:15: cannot use TxtAPI{} (type TxtAPI) as type APIInterfaces in return argument:
	TxtAPI does not implement APIInterfaces (missing CPULogCount method)
/home/xaionaro/go/pkg/mod/github.com/9elements/converged-security-suite/v2@v2.4.1-0.20210719144404-92b4758a5973/pkg/hwapi/cpu_blacklist.go:1027:14: t.ProcessorBrandName undefined (type TxtAPI has no field or method ProcessorBrandName)
/home/xaionaro/go/pkg/mod/github.com/9elements/converged-security-suite/v2@v2.4.1-0.20210719144404-92b4758a5973/pkg/hwapi/cpu_whitelist.go:796:14: t.ProcessorBrandName undefined (type TxtAPI has no field or method ProcessorBrandName)
```
```
$ go version
go version go1.16.6 linux/ppc64le
```
```
$ uname -a
Linux ubuntu 4.15.0-112-generic #113-Ubuntu SMP Thu Jul 9 23:41:09 UTC 2020 ppc64le ppc64le ppc64le GNU/Linux
```

xaionaro, 2021-07-27:
@ChriMarMe : Hey. Could somebody from your side take a look at this? I guess this problem will also be a blocker for @paulmenzel (see https://github.com/9elements/converged-security-suite/issues/268)

ChriMarMe, 2021-07-27:
You need to crosscompile for amd64.

`GOOS=Linux GOARCH=amd64 go get github.com/9elements/converged-security-suite/v2/cmd/pcr0tool@master`

ChriMarMe, 2021-07-27:
Or implement the "cpuid_ppc64.go" with dummy functions for the api

xaionaro, 2021-07-27:
> You need to crosscompile for amd64.
> 
> `GOOS=Linux GOARCH=amd64 go get github.com/9elements/converged-security-suite/v2/cmd/pcr0tool@master`

Well, it won't allow to execute the tool on PPC64LE :)

> Or implement the "`pkg/hwapu/cpuid_ppc64.go`" with dummy functions for the api

I guess this is a sign of a deeper problem. May be we should not have a dependency on hwapi at all in tools where we do not read anything through it.

ChriMarMe, 2021-07-27:
Go modules at its best....

ChriMarMe, 2021-07-27:
https://github.com/9elements/converged-security-suite/pull/270

closed 2021-07-27 by xaionaro