9elements/converged-security-suite #274: ppc64le: error: unable to analyze path './pkg/intel/metadata/manifest'

paulmenzel, 2021-07-29:
On ppc64le (Ubuntu 21.04 with go 1.16.2), cloning the repository, the command from the CircleCI configuration fails.

https://github.com/9elements/converged-security-suite/blob/f9990f16600821b0f9c9ec5d1f01c8dca7bb0a77/.circleci/config.yml#L16

```
$ git clean -dfx && git reset --hard
$ git describe --tags --dirty
v2.4.0-220-gf9990f1
$ GO111MODULE=off go get github.com/tjfoc/gmsm/sm2 github.com/dustin/go-humanize github.com/linuxboot/fiano/pkg/uefi && go run ./pkg/intel/metadata/manifest/common/manifestcodegen/cmd/manifestcodegen -check ./pkg/intel/metadata/manifest ./pkg/intel/metadata/manifest/bootpolicy ./pkg/intel/metadata/manifest/key
2021/07/29 10:35:38 an error: unable to analyze path './pkg/intel/metadata/manifest': unable to get the package from directory './pkg/intel/metadata/manifest': unable to open go directory: unable to normalize pkg path '': unable to find directory './pkg/intel/metadata/manifest' in paths [/usr/lib/go-1.16/src /home/pmenzel/go/src]
2021/07/29 10:35:38 an error: unable to analyze path './pkg/intel/metadata/manifest/bootpolicy': unable to get the package from directory './pkg/intel/metadata/manifest/bootpolicy': unable to open go directory: unable to normalize pkg path '': unable to find directory './pkg/intel/metadata/manifest/bootpolicy' in paths [/usr/lib/go-1.16/src /home/pmenzel/go/src]
2021/07/29 10:35:38 an error: unable to analyze path './pkg/intel/metadata/manifest/key': unable to get the package from directory './pkg/intel/metadata/manifest/key': unable to open go directory: unable to normalize pkg path '': unable to find directory './pkg/intel/metadata/manifest/key' in paths [/usr/lib/go-1.16/src /home/pmenzel/go/src]
exit status 1
$ git diff --stat
 pkg/intel/metadata/manifest/bootpolicy/Reserved_manifestcodegen.go           |  194 ------------
 pkg/intel/metadata/manifest/bootpolicy/bpmh_manifestcodegen.go               |  380 -----------------------
 pkg/intel/metadata/manifest/bootpolicy/manifest_manifestcodegen.go           |  448 ---------------------------
 pkg/intel/metadata/manifest/bootpolicy/pcd_manifestcodegen.go                |  240 ---------------
 pkg/intel/metadata/manifest/bootpolicy/pm_manifestcodegen.go                 |  246 ---------------
 pkg/intel/metadata/manifest/bootpolicy/se_manifestcodegen.go                 | 1055 ----------------------------------------------------------------
 pkg/intel/metadata/manifest/bootpolicy/signature_manifestcodegen.go          |  201 -------------
 pkg/intel/metadata/manifest/bootpolicy/txt_control_flags_manifestcodegen.go  |  138 ---------
 pkg/intel/metadata/manifest/bootpolicy/txt_manifestcodegen.go                |  664 ----------------------------------------
 pkg/intel/metadata/manifest/chipset_ac_module_information_manifestcodegen.go |  565 ----------------------------------
 pkg/intel/metadata/manifest/crypto_routines_manifestcodegen.go               |   43 ---
 pkg/intel/metadata/manifest/hash_manifestcodegen.go                          |  333 --------------------
 pkg/intel/metadata/manifest/key/hash_manifestcodegen.go                      |  180 -----------
 pkg/intel/metadata/manifest/key/manifest_manifestcodegen.go                  |  455 ----------------------------
 pkg/intel/metadata/manifest/key_manifestcodegen.go                           |  247 ---------------
 pkg/intel/metadata/manifest/key_signature_manifestcodegen.go                 |  202 -------------
 pkg/intel/metadata/manifest/signature_manifestcodegen.go                     |  252 ----------------
 pkg/intel/metadata/manifest/structure_manifestcodegen.go                     |  213 -------------
 pkg/intel/metadata/manifest/tpm_info_list_manifestcodegen.go                 |  269 -----------------
 19 files changed, 6325 deletions(-)
```

xaionaro, 2021-07-29:
Have you also considered this line? --
```
working_directory: /go/src/github.com/9elements/converged-security-suite
```

This command works only when you have the root of the project as the Current Working Directory.

paulmenzel, 2021-07-29:
Thank you for your answer. Unfortunately, I do not know what you mean. The pasted commands were executing in the top level of the converged-security-suite repository.

xaionaro, 2021-07-29:
Is the directory of `converged-security-suite` inside `go env GOPATH`?

paulmenzel, 2021-07-29:
> To better understand the situation: is the directory of `converged-security-suite` inside `GOPATH` (try to execute: `go env GOPATH`)?

    $ go env GOPATH
    /home/pmenzel/go
    $ export GOPATH=/dev/shm/converged-security-suite:$GOPATH
    $ go env GOPATH
    $GOPATH/go.mod exists but should not

> Also could
> 
> ```
> go run ./pkg/intel/metadata/manifest/common/manifestcodegen/cmd/manifestcodegen -check ./pkg/intel/metadata/manifest ./pkg/intel/metadata/manifest/bootpolicy ./pkg/intel/metadata/manifest/key 
> ```
> 
> be skipped in your case?

```
$ GO111MODULE=off go get github.com/tjfoc/gmsm/sm2 github.com/dustin/go-humanize github.com/linuxboot/fiano/pkg/uefi
$ golint -set_exit_status ./cmd/txt-suite
$ go mod download
$ go mod verify
all modules verified
$ CGO_ENABLED=0 go build -ldflags '-w -extldflags "-static"' -o txt-prov cmd/txt-prov/*.go
# github.com/9elements/converged-security-suite/v2/pkg/hwapi
pkg/hwapi/phys_le.go:16:5: memPaths redeclared in this block
        previous declaration at pkg/hwapi/phys_be.go:16:5
pkg/hwapi/phys_le.go:18:6: pathRead redeclared in this block
        previous declaration at pkg/hwapi/phys_be.go:18:52
pkg/hwapi/phys_le.go:31:6: selectDevMem redeclared in this block
        previous declaration at pkg/hwapi/phys_be.go:31:30
pkg/hwapi/phys_le.go:47:17: TxtAPI.ReadPhys redeclared in this block
        previous declaration at pkg/hwapi/phys_be.go:47:6
pkg/hwapi/phys_le.go:58:17: TxtAPI.ReadPhysBuf redeclared in this block
        previous declaration at pkg/hwapi/phys_be.go:58:6
pkg/hwapi/phys_le.go:76:6: pathWrite redeclared in this block
        previous declaration at pkg/hwapi/phys_be.go:76:53
pkg/hwapi/phys_le.go:91:17: TxtAPI.WritePhys redeclared in this block
        previous declaration at pkg/hwapi/phys_be.go:91:6
$ CGO_ENABLED=0 go build -ldflags '-w -extldflags "-static"' -o cbnt-prov cmd/cbnt-prov/*.go
# github.com/9elements/converged-security-suite/v2/pkg/hwapi
pkg/hwapi/phys_le.go:16:5: memPaths redeclared in this block
        previous declaration at pkg/hwapi/phys_be.go:16:5
pkg/hwapi/phys_le.go:18:6: pathRead redeclared in this block
        previous declaration at pkg/hwapi/phys_be.go:18:52
pkg/hwapi/phys_le.go:31:6: selectDevMem redeclared in this block
        previous declaration at pkg/hwapi/phys_be.go:31:30
pkg/hwapi/phys_le.go:47:17: TxtAPI.ReadPhys redeclared in this block
        previous declaration at pkg/hwapi/phys_be.go:47:6
pkg/hwapi/phys_le.go:58:17: TxtAPI.ReadPhysBuf redeclared in this block
        previous declaration at pkg/hwapi/phys_be.go:58:6
pkg/hwapi/phys_le.go:76:6: pathWrite redeclared in this block
        previous declaration at pkg/hwapi/phys_be.go:76:53
pkg/hwapi/phys_le.go:91:17: TxtAPI.WritePhys redeclared in this block
        previous declaration at pkg/hwapi/phys_be.go:91:6
$ go test ./pkg/amd/...
ok      github.com/9elements/converged-security-suite/v2/pkg/amd/manifest       0.003s
$ go test ./pkg/check/
?       github.com/9elements/converged-security-suite/v2/pkg/check      [no test files]
$ go test ./pkg/errors/
ok      github.com/9elements/converged-security-suite/v2/pkg/errors     0.005s
$ go test ./pkg/hwapi/
# github.com/9elements/converged-security-suite/v2/pkg/hwapi [github.com/9elements/converged-security-suite/v2/pkg/hwapi.test]
pkg/hwapi/phys_le.go:16:5: memPaths redeclared in this block
        previous declaration at pkg/hwapi/phys_be.go:16:5
pkg/hwapi/phys_le.go:18:6: pathRead redeclared in this block
        previous declaration at pkg/hwapi/phys_be.go:18:52
pkg/hwapi/phys_le.go:31:6: selectDevMem redeclared in this block
        previous declaration at pkg/hwapi/phys_be.go:31:30
pkg/hwapi/phys_le.go:47:17: TxtAPI.ReadPhys redeclared in this block
        previous declaration at pkg/hwapi/phys_be.go:47:6
pkg/hwapi/phys_le.go:58:17: TxtAPI.ReadPhysBuf redeclared in this block
        previous declaration at pkg/hwapi/phys_be.go:58:6
pkg/hwapi/phys_le.go:76:6: pathWrite redeclared in this block
        previous declaration at pkg/hwapi/phys_be.go:76:53
pkg/hwapi/phys_le.go:91:17: TxtAPI.WritePhys redeclared in this block
        previous declaration at pkg/hwapi/phys_be.go:91:6
FAIL    github.com/9elements/converged-security-suite/v2/pkg/hwapi [build failed]
FAIL
```

xaionaro, 2021-07-29:
@paulmenzel : This errors are caused by other problem. It looks we fixed https://github.com/9elements/converged-security-suite/issues/269 in a wrong way.

So feel free to just keep skipping the `go run` command. And about `pkg/hwapi/phys_le.go` I hope @ChriMarMe will help to handle :)

ChriMarMe, 2021-07-29:
#273 should fix this problem as here as well :)

xaionaro, 2021-07-29:
@zaolin : Could you please stamp https://github.com/9elements/converged-security-suite/pull/273 ? :)

xaionaro, 2021-07-29:
@paulmenzel : It looks like @zaolin merged the change, and it looks like [it works](https://app.circleci.com/pipelines/github/9elements/converged-security-suite/1293/workflows/75b38703-7fd4-49a9-966a-901056d5c2ea/jobs/2104). Could you try again?

ChriMarMe, 2021-07-29:
I'm not sure if i put the build flag in the right file though

paulmenzel, 2021-07-29:
Thank you. That fixed it for me. Thank you for the prompt support and fixes.

closed 2021-07-29 by paulmenzel