9elements/converged-security-suite #324: Lenovo X1 Carbon Gen9 test data

zaolin, 2022-04-05:
[test-data.zip](https://github.com/9elements/converged-security-suite/files/8413886/test-data.zip)

zaolin, 2022-04-05:
@xaionaro @rihter007  the firmware.bin is just the first 16 MB which includes FIT, BPM and KM

xaionaro, 2022-04-13:
Started the investigation in this branch: https://github.com/9elements/converged-security-suite/tree/support/lenovo_x1_carbon_gen9

Added some tooling there. But got stuck on the very first measurement. PCR0_DATA does not match what it should.

xaionaro, 2022-04-13:
Just for myself:
```
$ ./fittool show -f ~/Downloads/pcr0/firmware.bin
[sudo] password for xaionaro:
#   | Type                             | Address              | Size     | Version | Checksum valid  | Checksum
---------------------------------------------------------------------------------------------------------------
0   | FITHeaderEntry            (0x00) | 0x2020205f5449465f   | 15       | 0x0100  | true            | 165
1   | MicrocodeUpdateEntry      (0x01) | 0xffb81000           | 0        | 0x0100  | false           | 0
2   | MicrocodeUpdateEntry      (0x01) | 0xffb9d000           | 0        | 0x0100  | false           | 0
3   | MicrocodeUpdateEntry      (0x01) | 0xffbb9000           | 0        | 0x0100  | false           | 0
4   | MicrocodeUpdateEntry      (0x01) | 0xffbd5000           | 0        | 0x0100  | false           | 0
5   | SACM                      (0x02) | 0xffc40000           | 0        | 0x0100  | false           | 0
6   | BIOSStartupModuleEntry    (0x07) | 0xffc80000           | 24576    | 0x0100  | false           | 0
7   | BIOSStartupModuleEntry    (0x07) | 0xffce0000           | 36864    | 0x0100  | false           | 0
8   | BIOSStartupModuleEntry    (0x07) | 0xffd70000           | 102400   | 0x0100  | false           | 0
9   | BIOSStartupModuleEntry    (0x07) | 0xfff00000           | 4096     | 0x0100  | false           | 0
10  | BIOSStartupModuleEntry    (0x07) | 0xfff20000           | 12288    | 0x0100  | false           | 0
11  | BIOSStartupModuleEntry    (0x07) | 0xfff50000           | 45056    | 0x0100  | false           | 0
12  | TXTPolicyRecord           (0x0A) | 0x7a040100710070     | 0        | 0x0000  | false           | 0
13  | KeyManifestRecord         (0x0B) | 0xfff1b000           | 853      | 0x0100  | false           | 0
14  | BootPolicyManifestRecord  (0x0C) | 0xfff15000           | 1209     | 0x0100  | false           | 0
```

xaionaro, 2022-04-13:
EventLog replays into the final PCR0 value dumped from TPM itself, so let's just analyze EventLog:

---

```
#                   : 0
PCR index           : 0
Event Type          : 3
Hash Algorithm      : 4
Digest              : 0000000000000000000000000000000000000000
Data                : ([]uint8) (len=17 cap=17) {
    > 00000000  53 74 61 72 74 75 70 4c  6f 63 61 6c 69 74 79 00  |StartupLocality.|
    > 00000010  03                                                |.|
}
```
Standard initialization with locality 3.

---

```
#                   : 1
PCR index           : 0
Event Type          : 7
Hash Algorithm      : 4
Digest              : B46182FC7C3584C4355D8F4B11AE839DF90A392B
Data                : ([]uint8) (len=27 cap=27) {
    > 00000000  42 6f 6f 74 20 47 75 61  72 64 20 4d 65 61 73 75  |Boot Guard Measu|
    > 00000010  72 65 64 20 53 2d 43 52  54 4d 00                 |red S-CRTM.|
}
```
We expect it to be PCR0_DATA, but I cannot reproduce it using provided TXT Public Space.

---

```
#                   : 2
PCR index           : 0
Event Type          : 8
Hash Algorithm      : 4
Digest              : E6BECECF6165181092670F9E2AE24F81242CAD67
Data                : ([]uint8) (len=20 cap=20) {
    > 00000000  4e 00 33 00 32 00 45 00  54 00 37 00 35 00 57 00  |N.3.2.E.T.7.5.W.|
    > 00000010  20 00 00 00                                       | ...|
}
```
We expect it to be "PCD Vendor Version". Should be extractible pretty easy, but will check later.

---

```
#                   : 3
PCR index           : 0
Event Type          : 2147483656
Hash Algorithm      : 4
Digest              : 39961767AAD0DA9DE54635D5C50A05B62C0AAA8C
Data                : ([]uint8) (len=16 cap=16) {
    > 00000000  00 00 f0 ff 00 00 00 00  00 00 01 00 00 00 00 00  |................|
}
```
It looks like it measures 0xFFF00000-0xFFF10000. To be validated.

---

```
#                   : 4
PCR index           : 0
Event Type          : 2147483656
Hash Algorithm      : 4
Digest              : 989851F1373AA190029584E28529E434B0AFF628
Data                : ([]uint8) (len=16 cap=16) {
[... 128 lines cut ...]
---
```
#                   : 18
PCR index           : 0
Event Type          : 17
Hash Algorithm      : 4
Digest              : F07DF548DCA6DC4286EF4CE0D6DB12F02EA03E06
Data                : ([]uint8) (len=32 cap=32) {
    > 00000000  49 6e 74 65 6c 43 53 78  45 49 6e 66 6f 45 76 65  |IntelCSxEInfoEve|
    > 00000010  6e 74 00 00 01 00 00 00  86 80 e0 a0 01 00 00 00  |nt..............|
}
```
And I have no idea what is this.
---
```
#                   : 38
PCR index           : 0
Event Type          : 4
Hash Algorithm      : 4
Digest              : 9069CA78E7450A285173431B3E52C5C25299E473
Data                : ([]uint8) (len=4 cap=4) {
    > 00000000  00 00 00 00                                       |....|
}
```
Standard and well-known separator.

xaionaro, 2022-04-13:
OK, found an explanation for `IntelCSxE`

xaionaro, 2023-07-05:
> Started the investigation in this branch: https://github.com/9elements/converged-security-suite/tree/support/lenovo_x1_carbon_gen9
> 
> Added some tooling there. But got stuck on the very first measurement. PCR0_DATA does not match what it should.

Started a new branch: https://github.com/9elements/converged-security-suite/tree/feature/lenovo_x1_carbon_gen9