9elements/converged-security-suite #356: Issues following bg-prov instructions, may be several bugs.

65a, 2023-04-02:
Thanks for this project, it's awesome!

I am trying to write new bootguard metadata to a sapphire rapids board, and found this, which is perfect. I ran into a few issues.

I'm following use case 1 here: https://github.com/9elements/converged-security-suite/blob/main/cmd/bg-prov/README.md

First, `bg-prov template foo.cfg` doesn't exist, but I looked at the help output and found `bg-prov template-v-2 foo.cfg` should be right. I get a nil pointer dereference:
```
$ ./bg-prov template-v-2 ./bg2.cfg
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x5a01be]

goroutine 1 [running]:
github.com/linuxboot/fiano/pkg/intel/metadata/bg/bgbootpolicy.(*Manifest).WriteTo(0x0, {0x79af60, 0xc0002318f0})
	/home/user/go/pkg/mod/github.com/linuxboot/fiano@v1.1.4-0.20230131115913-85ddba13ba44/pkg/intel/metadata/bg/bgbootpolicy/manifest_manifestcodegen.go:209 +0x3e
github.com/9elements/converged-security-suite/v2/pkg/provisioning/bootguard.(*BootGuard).WriteBPM(0xc000231800)
	/home/user/devel/converged-security-suite/pkg/provisioning/bootguard/bootguard.go:331 +0x5a
main.(*templateCmdv2).Run(0x976ee0, 0x1?)
	/home/user/devel/converged-security-suite/cmd/bg-prov/cmd.go:910 +0x3d3
reflect.Value.call({0x6d2ae0?, 0x976ee0?, 0x44d1d6?}, {0x722762, 0x4}, {0xc000013b48, 0x1, 0x1?})
	/usr/lib/go/src/reflect/value.go:586 +0xb07
reflect.Value.Call({0x6d2ae0?, 0x976ee0?, 0x6dd540?}, {0xc000013b48?, 0x721240?, 0x0?})
	/usr/lib/go/src/reflect/value.go:370 +0xbc
github.com/alecthomas/kong.callMethod({0x72256c, 0x3}, {0x719760?, 0x976ee0?, 0x3?}, {0x6d2ae0?, 0x976ee0?, 0x0?}, 0x0?)
	/home/user/go/pkg/mod/github.com/alecthomas/kong@v0.7.1/callbacks.go:95 +0x4fa
github.com/alecthomas/kong.(*Context).RunNode(0xc00011a600, 0xc000167680, {0xc000125f00, 0x1, 0x1})
	/home/user/go/pkg/mod/github.com/alecthomas/kong@v0.7.1/context.go:755 +0x60f
github.com/alecthomas/kong.(*Context).Run(0x6b83e0?, {0xc000125f00?, 0xc000125f30?, 0x4408b1?})
	/home/user/go/pkg/mod/github.com/alecthomas/kong@v0.7.1/context.go:780 +0x14e
main.main()
	/home/user/devel/converged-security-suite/cmd/bg-prov/main.go:31 +0x29e
```

I can `read-config`, so I generated a config.json from an existing image. You can take a publically available image from SuperMicro for example, but I suspect any Sapphire Rapids (or W790?) image will suffice. This works, so I keep following the steps.

I get as far as:
`/bg-prov bpm-gen-v-2 ./bpm_unsigned.bin ./oem_bios.bin --config=./oem.cfg`
which just gives me
```
can't identify bootguard header
WriteBPM: can't identify bootguard header
```
I'm new to this, so I may be doing something terribly wrong here. Let's assume the fuses are not locked in the ME, so replacing keys here *should* be ok if I understand correctly. I'd like to resign an existing BIOS with my own keys.

xaionaro, 2023-04-05:
Hello.

Do you mind sharing the files you used to reproduce this problem? Or otherwise could you try branch `bugfix/cbnt-prov-typo` and check if it works?

65a, 2023-04-06:
You can download an example BIOS at https://www.supermicro.com/en/support/resources/downloadcenter/firmware/MBD-X13SEM-TF/BIOS (I suspect all X13 LGA4677 boards will have the same issues). I see there is an SPR-SP branch too, which is probably needed as well for these boards.  

Running bugfix branch, this problem seems unrelated to your change (codegen?):
```
$ ./bg-prov template-v-2 test
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x5a01be]

goroutine 1 [running]:
github.com/linuxboot/fiano/pkg/intel/metadata/bg/bgbootpolicy.(*Manifest).WriteTo(0x0, {0x79af60, 0xc0002318f0})
	/home/user/go/pkg/mod/github.com/linuxboot/fiano@v1.1.4-0.20230131115913-85ddba13ba44/pkg/intel/metadata/bg/bgbootpolicy/manifest_manifestcodegen.go:209 +0x3e
github.com/9elements/converged-security-suite/v2/pkg/provisioning/bootguard.(*BootGuard).WriteBPM(0xc000231800)
	/home/user/devel/converged-security-suite/pkg/provisioning/bootguard/bootguard.go:331 +0x5a
main.(*templateCmdv2).Run(0x976e40, 0x1?)
	/home/user/devel/converged-security-suite/cmd/bg-prov/cmd.go:910 +0x3d3
reflect.Value.call({0x6d2ae0?, 0x976e40?, 0x44d1d6?}, {0x722762, 0x4}, {0xc000013b48, 0x1, 0x1?})
	/usr/lib/go/src/reflect/value.go:586 +0xb07
reflect.Value.Call({0x6d2ae0?, 0x976e40?, 0x6dd540?}, {0xc000013b48?, 0x721240?, 0x0?})
	/usr/lib/go/src/reflect/value.go:370 +0xbc
github.com/alecthomas/kong.callMethod({0x72256c, 0x3}, {0x719760?, 0x976e40?, 0x3?}, {0x6d2ae0?, 0x976e40?, 0x0?}, 0x0?)
	/home/user/go/pkg/mod/github.com/alecthomas/kong@v0.7.1/callbacks.go:95 +0x4fa
github.com/alecthomas/kong.(*Context).RunNode(0xc00011a600, 0xc000167680, {0xc000125f00, 0x1, 0x1})
	/home/user/go/pkg/mod/github.com/alecthomas/kong@v0.7.1/context.go:755 +0x60f
github.com/alecthomas/kong.(*Context).Run(0x6b83e0?, {0xc000125f00?, 0xc000125f30?, 0x4408b1?})
	/home/user/go/pkg/mod/github.com/alecthomas/kong@v0.7.1/context.go:780 +0x14e
main.main()
	/home/user/devel/converged-security-suite/cmd/bg-prov/main.go:31 +0x29e
```
Continuing on:
`./bg-prov read-config test bios.bin` works fine, contents of `test` seem legitimate.
`./bg-prov key-gen RSA3072 "" --path=sign` works fine
This is a new problem, I think:
```
$ ./bg-prov km-gen-v-2 ./km_unsigned.bin signkm_priv.pem --config test --pkhashalg SHA384 --bpmpubkey signbpm_pub.pem --bpmhashalgo SHA384
bg-prov: error: asn1: structure error: tags don't match (16 vs {class:0 tag:2 length:1 isCompound:false}) {optional:false explicit:false application:false private:false defaultValue:<nil> tag:<nil> stringType:0 timeType:0 set:false omitEmpty:false} AlgorithmIdentifier @2
```
Note that trying the above with 2048 keys/algo 14 also fails with a similar message.

xaionaro, 2023-04-06:
```
github.com/linuxboot/fiano/pkg/intel/metadata/bg/bgbootpolicy.(*Manifest).WriteTo(0x0, {0x79af60, 0xc0002318f0})
	/home/user/go/pkg/mod/github.com/linuxboot/fiano@v1.1.4-0.20230131115913-85ddba13ba44/pkg/intel/metadata/bg/bgbootpolicy/manifest_manifestcodegen.go:209 +0x3e
github.com/9elements/converged-security-suite/v2/pkg/provisioning/bootguard.(*BootGuard).WriteBPM(0xc000231800)
	/home/user/devel/converged-security-suite/pkg/provisioning/bootguard/bootguard.go:331 +0x5a
main.(*templateCmdv2).Run(0x976e40, 0x1?)
```
Does not look right. Are you sure you are working from that branch? In that branch it should get to `pkg/intel/metadata/cbnt` instead of `pkg/intel/metadata/bg` from `pkg/provisioning/bootguard/bootguard.go:331`.

I hope I won't forget to investigate this on the next week :(

ansiwen, 2023-08-10:
I have exactly the same issue on main branch. Also template-v-1 does not create a JSON file, but a binary, and looking at the code it seems that the template-v-1 and template-v-2 commands indeed don't create JSON files, but BPM files. See here: https://github.com/9elements/converged-security-suite/blob/d249aa65ddf9e30b532e7a785b0f902e8a520646/cmd/bg-prov/cmd.go#L949-L955

65a, 2023-08-11:
I'm definitely still interested in getting this project working for SPR, though I ran into some other annoyances with Supermicro particularly DRMing their board to their own keys with an external FPGA (something they call RoT, but I might dispute the T). This may be bypassable, but I have another board which is hopefully less annoying. Let me know if there's anything I can test, I'll likely try again soon.

zaolin, 2023-09-05:
Related to #355